swoboda1337/sx127x-esphome #1: ESp32 with sx127x reciever

v12345vtm, 2024-09-06:
Hello , 
i don't know if i can mail to you directly , 
but your code works like a charm!
thanks for sharing it.

it works so good i want to add a RF keyfob.
i checked it freq via a rtlsdr and i see some rf keyfobs work at 433920000  and some at 433721000 

is there a solution to recieve more freq in the same setup ??

kind regard 
keep on the good work for esphome 

thank you

```
sx127x:
  nss_pin: GPIO18
  rst_pin: GPIO23
  frequency: 433721000 # 433920000 = deurbel   **** tvh = 433721000
  modulation: OOK
  rx_floor: -90
  rx_bandwidth: 50_0kHz

remote_receiver: #A proper antenna like the Siretta Tango 9 has better range and doesn’t glitch when Wi-Fi transmits.
  pin: GPIO32
  #filter: 255us
  #idle: 2000us
  #buffer_size: 100000b
  memory_blocks: 3 #8
  dump: 
    - rc_switch
    - raw

binary_sensor
  - platform: remote_receiver
    name: "tvh"
    raw:
      #code: [274, -666, 261, -653, 261, -667, 261, -653, 274, -654, 261, -665, 262, -653, 261, -666, 262, -666, 261, -653, 261, -666, 261, -667, 261, -653, 261, -666, 262, -665, 262, -391, 523, -405, 522, -654, 274, -653, 261, -405, 522, -392, 536 ]
      code : [  1803, -261, 1789, -262, 1789, -261, 1790, -274, 1776, -261, 1790, -261, 1789, -262, 1789, -274, 1790, -260, 1777, -313, 1738, -274, 1789, -261, 1777, -274, 1777, -274, 1789, -261, 1776 ]
    filters:
      - delayed_off: 500ms   

```

swoboda1337, 2024-09-06:
You could change the frequency in a lambda:

id(sx127x_id).set_frequency(433721000);
id(sx127x_id).setup();

It is a bit hacky that you have to call setup again. I do plan to get this driver officially merged into esphome. Before I do that I will make sure there is a better way to change settings without having to calling setup again.

swoboda1337, 2024-09-10:
I added a configure function, can change any setting and run configure:

id(sx127x_id).set_frequency(433721000);
id(sx127x_id).configure();

closed 2024-09-10 by swoboda1337