swoboda1337/sx127x-esphome #5: 2 LILYGO TTGO LoraV2.1-1.6 to communicate

paradisgaston, 2025-02-06:
Hi, I am new to this type of communication.
I purchased two LILYGO TTGO LoraV2.1-1.6 that I which to exchange text at a distance of 1km.
I configure the 2 devices using your external _components github://swoboda1337/sx127x-esphome.
I change the frequency to 915000000  # 915 MHz in Hz because I wish to use LoRa. And -sx127x.send_packet: "Hello Cabane" instead of raw.

Does your external_compnent support this frequency and sending data. I have define button in each device to send message. But I do not see anything happening.
If you wish, I can send you my code to investigate. 
Thanks

swoboda1337, 2025-02-06:
The two devices are close together now or 1 km apart?

Can look at the packet mode docs here: https://deploy-preview-4278--esphome.netlify.app/components/sx127x#packet-mode

That config is using FSK and probably has a range of 500 meters. 

I haven't added support for LoRa yet but it is on my todo list.

swoboda1337, 2025-02-06:
As for frequency support yes it will support that frequency as long as you bought a device that supports that frequency.

paradisgaston, 2025-02-06:
Wow, this is quick. My 2 devices are about 1 meter of each other. My
purchased model is LILYGO 868/915MHz.

I will check the documentation that you recommend

Gaston

paradisgaston, 2025-02-06:
Hi Jonathan, being totally new to this LoRa communication and working with
ChapGPT I came out with this code
external_components:
  - source: github://swoboda1337/sx127x-esphome

sx127x:     # LoRa Configuration
  nss_pin: GPIO18
  rst_pin: GPIO23
  modulation: OOK
  frequency: 915000000  # 915 MHz in Hz
  pa_power: 17
  on_packet:
    then:
      - logger.log: "Received LoRa message: ${data}"
      - lambda: |-
          ESP_LOGD("lambda", "packet %s", format_hex(x).c_str());

button:
  - platform: template
    name: "Send Message"
    on_press:
      then:
        - sx127x.send_packet: "Hello Cabane"
        - delay: 100ms  # Small delay to ensure the packet is sent
        - sx127x.set_mode_standby  # Set the module back to standby mode A
lot less line of code, major difference is frequency ans send_packet
instead of raw. I will try you config and let you know what happen Thank
you,

swoboda1337, 2025-02-06:
Just use the config in that link exactly as is but with a different frequency:

```
# Example configuration entry
sx127x:
  dio0_pin: GPIO26
  nss_pin: GPIO18
  rst_pin: GPIO23
  pa_pin: BOOST
  pa_power: 17
  bitsync: true
  bitrate: 4800
  frequency: 915000000  
  modulation: FSK
  rx_start: true
  payload_length: 8
  sync_value: [0x33, 0x33]
  preamble_size: 2
  preamble_errors: 8
  preamble_polarity: 0x55
  on_packet:
    then:
      - lambda: |-
          ESP_LOGD("lambda", "packet %s", format_hex(x).c_str());

  button:
    - platform: template
      name: "Transmit Packet"
      on_press:
        then:
          - sx127x.send_packet:
             data: [0xC5, 0x51, 0x78, 0x82, 0xB7, 0xF9, 0x9C, 0x5C
```

paradisgaston, 2025-02-06:
I change my config and I am receiving packages
TimelevelTagMessage
16:01:23 [D] [lambda:072]

packet 327856adba30fb3f

16:02:07 [D] [lambda:076]

packet 317856adba31fb3f

If this is 1 and a 2 SUCCESS. Would you know how to convert this to
readable text ?

swoboda1337, 2025-02-07:
Looks like its working. To do text you can do:
```
        - sx127x.send_packet:
            data: !lambda |-
              std::string message = "Hello!";
              std::vector<uint8_t> data(message.begin(), message.end());
              data.resize(8);
              return data;
```
and 
```
  on_packet:
    then:
      - lambda: |-
          std::string message(x.begin(), x.end());
          ESP_LOGD("lambda", "packet %s", message.c_str());
```
Note the packet size is fixed so I put a resize 8 there (change that if you increase the packet size)

paradisgaston, 2025-02-07:
Great, I got my first messages "hello" and "12345678".
So the length is fix ? How long can I make it ?|
I will have to transmit 2 temp 20.0 20.0 and a water level which will be in
integer 1-4, a boolean to trigger a relay and finally a timestamp (or
hour:minute)

swoboda1337, 2025-02-07:
Max size is 64. It would be more efficient to send thing as bytes not text.

paradisgaston, 2025-02-11:
Hi Jonathan, my program works perfectly thanks to you.
One clarification question :
 rx_start:           true          # Start the receiver automatically on boot.   (Receiver configuration variables).
If I set it to false, I will have to start RX first time I send a packet or every time I send a packet ?
H