swoboda1337/sx127x-esphome #6: Packet and Continuous Mode. Payload_length

paradisgaston, 2025-02-16:
Hi Jonathan, you work fast on this one. You just gave me the code last week, that work perfectly.
Question, can you explain what is the difference between packet and continuous mode ? I am using packet mode presently. Advantages of both options.

Is there a way to send and received different packet size. Currently, I am sending and receiving packet size 12, which contains : originator, destination, packet_type …. Plus a bunch of sensor value float and integer. 
I would like to also use a small pack (originator, destination, packet_type and 1 extra byte. For example, I would command the remote station to turn on a relay to switch à light. Or could be a confirmation for received message.

swoboda1337, 2025-02-16:
I added LoRa support which supports variable length packets up to 256 bytes. That is working and available here:
```
external_components:
  - source: github://swoboda1337/sx127x-esphome@main
    components: [ sx127x ]
```
I need a day or two to get the docs and the PR updated. Something like this would work:
```
sx127x:
  dio0_pin: GPIO26
  cs_pin: GPIO18
  rst_pin: GPIO23
  pa_pin: BOOST
  pa_power: 17
  bandwidth: 125_0kHz
  crc_enable: true
  frequency: 433920000
  modulation: LORA
  code_rate: 4_6
  spreading_factor: 7
  rx_start: true
```
Can play around with the bandwidth, spreading_factor and code_rate.

swoboda1337, 2025-02-16:
Right. Continuous mode is mostly used for OOK modulation. If you want to do esp to esp I would not recommend it

swoboda1337, 2025-02-25:
Max packet size is 256 for LoRa. Yeah there is no way to automatically break things up and reassemble them. Shouldn't be too hard in a lambda but could also write a separate component to do that.

baxgit, 2025-02-26:
OK, thanks for confirming. Is there a 'recommended' way to check the packet size?

swoboda1337, 2025-02-26:
Of the received packet? You can check x.size() 

If you mean transmit if you set the payload_length to zero you can send any size you want up to 256