swoboda1337/sx127x-esphome #7: Configure SX1276 on LilyGO LoRa32 v2.1_1.6 for WMBus Data Reception at 868 MHz

johnyvmware, 2025-03-18:
Hello,

I'm trying to configure the SX1276 on a LilyGO LoRa32 v2.1_1.6 board to receive WMBus data on 868 MHz. I have been using the repository's examples and tried configuring the module in both continuous and packet modes with FSK modulation, but unfortunately, I cannot get it to log any received data.

Here are the configurations I've tried:

**Continuous Mode Configuration:**
```
sx127x:
  id: sx127x_radio
  cs_pin: GPIO18
  rst_pin: GPIO23
  frequency: 868950000
  modulation: FSK
  rx_start: true
  bandwidth: 50_0kHz
  rx_floor: -95

remote_receiver:
  pin: GPIO32
  dump: raw
```

**Packet Mode Configuration:**
```
sx127x:
  cs_pin: GPIO18
  rst_pin: GPIO23
  dio0_pin: GPIO26
  pa_pin: BOOST
  pa_power: 17
  bitrate: 4800
  bitsync: true
  crc_enable: true
  frequency: 868950000
  modulation: FSK
  rx_start: true
  payload_length: 8
  sync_value: [0x33, 0x33]
  preamble_size: 2
  preamble_errors: 8
  preamble_polarity: 0x55
  on_packet:
    then:
      - lambda: |-
          ESP_LOGD("lambda", "packet %s", format_hex(x).c_str());
```

In both modes, no packets are being logged. My ultimate goal is to receive raw WMBus data, and later on, I'll introduce decoding.

I would greatly appreciate any advice, examples, or modifications to the configurations that could help me get this working.

Thank you in advance for your assistance!

swoboda1337, 2025-03-18:
Is this the correct protocol here: https://github.com/merbanan/rtl_433/blob/da289d37eba7378726fcde502d38a466f9128b22/src/devices/m_bus.c#L855

I may have to add another config to enable manchester encoding.  You can see the preamble is 0x55, 0x54, 0x76, 0x96. Would need only allow 1 preamble byte and set the sync to 0x54, 0x76x 0x96. 

I can look at adding a manchester option tonight or tomorrow.

swoboda1337, 2025-03-18:
Also disable crc as you dont know the length of the packet or the polynomial.

johnyvmware, 2025-03-18:
Thank you for your response and for looking into this!

The provided protocol in the link looks correct, and the clarification about the preamble sequence is very helpful. I’ll try the updated configuration with the adjusted sync value and preamble settings. Disabling CRC also makes sense, given the unknown polynomial and packet length.

I’ll test these changes and report back on the results.

Really appreciate your help!

swoboda1337, 2025-03-18:
Also note the frequency in that link is different: Frequency 868.33 MHz, Bitrate 4.8 kbps

johnyvmware, 2025-03-19:
I tried your suggestion and was able to see data being logged in the log output—thank you for pointing me in the right direction! Turns out my electric meter uses Mode T, so I updated the frequency, bitrate, and related configurations. My current setup looks like this:

```
sx127x:
  cs_pin: GPIO18
  rst_pin: GPIO23
  dio0_pin: GPIO26
  pa_pin: BOOST
  pa_power: 17
  bitrate: 100000
  bitsync: true
  crc_enable: false
  frequency: 868950000
  modulation: FSK
  rx_start: true
  payload_length: 8
  sync_value: [0x54, 0x3D] # Mode C&T from m_bus_c_t_callback
  preamble_size: 0
  preamble_detect: 0
  on_packet:
    then:
      - lambda: |-
          ESP_LOGD("lambda", "packet %s", format_hex(x).c_str());
```

However, I have a few doubts. The [rtl_433 Mode T implementation](https://github.com/merbanan/rtl_433/blob/da289d37eba7378726fcde502d38a466f9128b22/src/devices/m_bus.c#L935C1-L936C1) mentions a preamble of `{0x54, 0x3D}`. I’ve set this as the `sync_value` and disabled `preamble_size` and `preamble_detect` by setting them to `0`. Does this approach seem correct? I observed signals when using `preamble_polarity` values of both `0x55` and `0xAA`. Should this affect the reception, or is it safe to leave it as-is?

Thanks again for your help and insights!

swoboda1337, 2025-03-19:
The rtl 433 seems to say the bitrate should be 32768, the frequency 868.3. For the preamble I am not sure as this is manchester encoded, lets remove the sync altogether for now. For now can try:
```
sx127x:
  cs_pin: GPIO18
  rst_pin: GPIO23
  dio0_pin: GPIO26
  pa_pin: BOOST
  pa_power: 17
  bitrate: 32768
  bitsync: true
  crc_enable: false
  frequency: 868300000
  modulation: FSK
  rx_start: true
  payload_length: 8
  # sync_value: [0x54, 0x3D] # Mode C&T from m_bus_c_t_callback
  preamble_size: 0
  preamble_detect: 0
  on_packet:
    then:
      - lambda: |-
          ESP_LOGD("lambda", "packet %s", format_hex(x).c_str());
```

swoboda1337, 2025-03-20:
Try with and without the sync:

```
sx127x:
  cs_pin: GPIO18
  rst_pin: GPIO23
  dio0_pin: GPIO26
  pa_pin: BOOST
  pa_power: 17
  bitrate: 32768
  bitsync: true
  crc_enable: false
  frequency: 868300000
  modulation: FSK
  rx_start: true
  payload_length: 8
  sync_value: [0x54, 0x3D] # Mode C&T from m_bus_c_t_callback
  preamble_size: 0
  preamble_detect: 0
  on_packet:
    then:
      - lambda: |-
          ESP_LOGD("lambda", "packet %s", format_hex(x).c_str());
```

swoboda1337, 2025-03-20:
Actually let me google this a bit more tonight. Its not clear exactly what it should look like.

swoboda1337, 2025-03-20:
Ok this document was very helpful: https://www.infineon.com/dgdl/TDA5340_AN_WMBus_v1.0.pdf

[image]

For T1 and T2 there is 19 x 01 and then 0000111101. That means we can look for two preamble bytes then 0x54 0x3D. It does seem the bit rate is 100kbits and the frequency is 868.95. So yeah your config is almost right. Lets add the preamble back:

```
sx127x:
  cs_pin: GPIO18
  rst_pin: GPIO23
  dio0_pin: GPIO26
  pa_pin: BOOST
  pa_power: 17
  bitrate: 100000
  bitsync: true
  crc_enable: false
  frequency: 868950000
  modulation: FSK
  rx_start: true
  payload_length: 8
  sync_value: [0x54, 0x3D] # Mode C&T from m_bus_c_t_callback
  preamble_size: 2
  preamble_detect: 2
  preamble_errors: 8
  preamble_polarity: 0x55
  on_packet:
    then:
      - lambda: |-
          ESP_LOGD("lambda", "packet %s", format_hex(x).c_str());
```

swoboda1337, 2025-03-20:
You can also try with `bandwidth: 200_0kHz`

johnyvmware, 2025-03-20:
First off, thank you for sharing the Infineon document - it has been incredibly helpful in demystifying WMBus operations.

I do have a question regarding Mode T requirements. For `n >= 19`, where `n x (01)` is expected, is it necessary to configure 4 bytes as `0x55`, followed by the sync values `0x54` and `0x3D`, to achieve the required synchronization sequence? If that’s the case, would the current code limitation - allowing for a maximum preamble length of 3 bytes - need to be addressed? For reference, here’s [the relevant code section](https://github.com/swoboda1337/sx127x-esphome/blob/53a6b90d377fbab8835a165e6128bd138395903c/components/sx127x/__init__.py#L205C1-L206C1).

Additionally, while reviewing the documentation, I noticed that the length of a WMBus message is stated to be included in the first byte. According to the docs, `payload_length` is required for packet mode, with a maximum value of 64 bytes. While th code appears to set the limit to 64 bytes, the comment suggests "Payload length must be >= 64 with FSK/OOK." See [here](https://github.com/swoboda1337/sx127x-esphome/blob/53a6b90d377fbab8835a165e6128bd138395903c/components/sx127x/__init__.py#L163-L164). I'm wondering whether the component should support messages exceeding 64 bytes?

Finally, I came across Semtech’s SX1276 reference implementation ([LoRaMac-node](https://github.com/Lora-net/LoRaMac-node)), which indicates that the SX1276 can be configured for variable-length messages, where the length is determined automatically from the first byte. Here’s the [relevant code section](https://github.com/Lora-net/LoRaMac-node/blob/dcbcfb329b4a343ab007bc19ac43a8dc952b3354/src/radio/sx1276/sx1276.h#L247). Would you consider supporting such a mode in your implementation?

Thanks again for your time and support—it’s greatly appreciated.

swoboda1337, 2025-03-20:
For the preamble the hw needs that to sync frequency and timing. The recommended value is two bytes. Its ok for the signal to have a much longer preamble, could have 1000 bytes its ok. Once it finds a preamble it looks for the sync bytes to find the start of the packet. 

That error message should say "Payload length must be <= 64 with FSK/OOK" I will fix that, thanks. The hw buffer is 64 bytes long it would be a big change to stream that out in realtime. 

I could add variable-length messages, it wont be too hard. For now its easy enough to just to set a large size like 32 or something. Doesn't matter if you make it longer than the actual packet.

johnyvmware, 2025-03-20:
Thank you for the clarification regarding the preamble size. I understand it now.

As for the payload length, if I set it to 32 bytes, the incoming data array would also have a length of 32 bytes. From there, I would read the first byte to determine the actual length of the signal and then only process that part of the message. Is that correct?

closed 2025-03-21 by johnyvmware