swoboda1337/sx127x-esphome #11: Common interface/base class for LoRa chips

yawor, 2025-04-02:
Hi,

I've seen that you also have another repo for sx126x chips. I've been thinking that it would be nice to have a base component class for for sx126x and sx127x (and also potentially many other chips) which would provide a common interface for them. This may not be that important if all you do is using packet payload directly in `on_packet` event in yaml, as the yaml definition doesn't care about concrete classes that are then generated, but given that #10 is now merged and it gives the ability to use sx127x from another component, the other component is now bound to specific chip class. Even if the same code listener code is added to sx126x, it can't be used without changing the other component.

So I would propose something like `LoRaRadio` or `PacketRadio` class together with a `LoRaRadioListener` (or `PacketRadioListener`) classes with some sensible common methods like `register_listener`, `transmit_packet` and maybe even `set_mode_*` as virtual methods, which would be overridden by each implementation.

Then some other component would just need to inherit from a common listener class which could be registered to the radio component no matter which chip is used.

swoboda1337, 2025-04-02:
Yeah I agree a common component and a common interface would be ideal. 

My concern atm is I am trying to get these components merged into ESPHome. Creating a third component and a new interface for all radio will make it a lot harder to get these in. I am thinking it would be best to get these in first an then try to create a common interface?

yawor, 2025-04-02:
Yes, this can definitely wait. It's more a quality of life thing than a necessity for me. At least for me, for my component, I don't really need it to be compatible with different chips right now. It's to get data from quite niche, battery operated microwave fluid level sensors for septic tanks or rain water tanks etc, made by a Polish company. I don't even know if anyone would even be interested to use my component :P. I'm doing it mainly for myself, but I'll publish it when ready anyway.